thunderace/Esp8266-Arduino-Makefile #6: Compiling C as C++ breaks linker provides

chuckwagoncomputing, 2016-12-23:
Maybe I'm doing something wrong, but I had to comment out the CXX compile line and decomment the CC line to prevent errors.
I'm using code from [cnlohr's ws2812 project](https://github.com/cnlohr/esp8266ws2812i2s) which uses rom_i2c_writeReg_Mask, which is a symbol defined in eagle.rom.addr.v6.ld.
Compiling as C++ broke that somehow. I'm not clear on how the linker and compiler work together there, maybe someone else could explain it.
I suppose it would be possible to use extern, but upstream esp8266_peri.h needs rom_i2c_writeReg_Mask as well, so for me the best solution was to compile C code with regular old gcc.

thunderace, 2016-12-27:
Do you have a full project for tests?

chuckwagoncomputing, 2016-12-27:
Here's a MNWE.

```#ifndef i2c_bbpll
#define i2c_bbpll                                 0x67
#define i2c_bbpll_en_audio_clock_out            4
#define i2c_bbpll_en_audio_clock_out_msb        7
#define i2c_bbpll_en_audio_clock_out_lsb        7
#define i2c_bbpll_hostid                           4

#define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
#define i2c_writeReg_Mask_def(block, reg_add, indata) \
      i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
#endif

void setup() {
 i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1);
}

void loop() {}```

thunderace, 2016-12-28:
You're right.
The last commit restore c files comilation with gcc.
But, if your code is located in a .ino file, the problem is still here.

Closed.

closed 2016-12-28 by thunderace