thunderace/Esp8266-Arduino-Makefile #7: Doesn't include libraries without a .cpp file

chuckwagoncomputing, 2016-12-24:
Some libraries, like Servo, have no <name>.cpp file in their src directory, only <name>.h.
These libraries are not found.

thunderace, 2016-12-27:
It's because with Servo Library, the source code is under src/esp8266...
Try to add these 3 lines in esp8266Arduino.mk at line 108:
	$(ARDUINO_LIBS:%=$(ARDUINO_HOME)/libraries/%/src/*/*.c) \
	$(ARDUINO_LIBS:%=$(ARDUINO_HOME)/libraries/%/src/*/*.cpp) \
	$(ARDUINO_LIBS:%=$(ARDUINO_HOME)/libraries/%/src/*.h) \
tested with Servo library

chuckwagoncomputing, 2016-12-27:
Works fine now. I'd already made the modification before you suggested it, but I cloned again to test code for #6, so I went ahead and tried using the servo library. No errors, as expected.

closed 2016-12-28 by thunderace