thunderace/Esp8266-Arduino-Makefile #9: xtensa-lx106-elf-gcc: Command not found

mat-cab, 2017-07-09:
After a nominal install on Debian, I get the following error 
make: /home/me/Esp8266-Arduino-Makefile/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc : command not found

However when i look in the /home/me/Esp8266-Arduino-Makefile/xtensa-lx106-elf/bin directory, the file is present, with appropriate permissions (-rwxr-xr-x). Manual try to execute the (./xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc) fails with the same error (command not found).

Manually installing xtensa-lx106-elf-gcc (through esp-open-sdk) and esptool (through esptool-ck) and modifying the Makefile script to point to the appropriate location fixes that...

thunderace, 2017-07-13:
Strange... have you use the right install script (install-x86_64-pc-linux-gnu.sh for debian 64 or install-i686-pc-linux-gnu.sh for debian 32)?

dronov, 2017-07-19:
@thunderace install script changed to esp8266-install.sh, I also have this problem building my docker image for this purposes.

thunderace, 2017-07-19:
Hello,
I think that your problem is elsewhere not related to the makefile

Can you run this command : make print-CC and make print-LD

Regards.

dronov, 2017-07-19:
@thunderace hey and thanks for your work. 
With new script I have this error on Debian (while building docker container) and Ubuntu (on host machine) at installation
```   creating: esp8266-2.3.0/variants/wifinfo/
  inflating: esp8266-2.3.0/variants/wifinfo/pins_arduino.h  
   creating: esp8266-2.3.0/variants/wifio/
  inflating: esp8266-2.3.0/variants/wifio/pins_arduino.h  
  inflating: esp8266-2.3.0/variants/wifio/WifioProtocol.h  
  inflating: esp8266-2.3.0/variants/wifio/WifioWiring.cpp  
Platform: x86_64-pc-linux-gnu
Traceback (most recent call last):
  File "./get.py", line 113, in <module>
    tools_to_download = load_tools_list('../package/package_esp8266com_index.template.json', identify_platform())
  File "./get.py", line 87, in load_tools_list
    tools_info = json.load(open(filename))['packages'][0]['tools']
IOError: [Errno 2] No such file or directory: '../package/package_esp8266com_index.template.json'
```
Compiler did not downloaded. If I checked out to previous (e.g. December) commits and x86_64 script -- it works well.

dronov, 2017-07-19:
@thunderace here it is my work-in-progress Dockerfile:
https://paste.gnome.org/py5dzf7cm

thunderace, 2017-07-19:
you are right, there is no package directory in the 2.3.0 archive.
It's strange because in my repository, i have it in my old installation (06/23/2016).

The last commit fix this. Update your repository and run esp8266-install.sh again

dronov, 2017-07-19:
@thunderace check this out please :)
https://github.com/thunderace/Esp8266-Arduino-Makefile/pull/10

thunderace, 2017-07-19:
thx dronov. Fixed and PR #10 closed

thunderace, 2017-07-19:
I have no problem with your second dockerfile (https://paste.gnome.org/plo1twfd0)...

dronov, 2017-07-19:
@thunderace yeah, it works.

mat-cab, 2017-07-19:
The issue can be closed, I think I used x64 files instead of x32.

closed 2017-07-19 by mat-cab