thunderace/Esp8266-Arduino-Makefile #15: Different Version variables

stritti, 2017-12-19:
There are two version defines of Version:

# `ESP8266_VERSION` within `Makefile`/`espXArduino.mk`
# `ESP8266_VER` within `esp8266-install.sh`

I think, they are not synchonized. Maybe I am using it in wrong way, but I had to update both places.

thunderace, 2017-12-19:
The ESP8266_VER  within install8266-install.sh is the esp8266-Arduino last stable version number.
The ESP8266_VERSION (within Makefile) is the version used to generate the binary.
For example, install both the stable version (with esp8266_install.sh) and the git version (with esp8266-install-from-git). If you want to use the git version you just have to set this in your Makefile : ESP8266_VERSION=.git See the example/esp8266/AdvancedWebServer/Makefile content.

stritti, 2017-12-19:
Okay, I see. Thanks for introduction. 
Probably it would be valuable to describe the different usage within readme.

thunderace, 2017-12-19:
Yes, i will enhance the README before merge the dev branch in the master one.

closed 2017-12-19 by thunderace