thunderace/Esp8266-Arduino-Makefile #16: IRAM overflow

swapnil823, 2018-01-25:
IRAM Overflows while compiling [https://github.com/armtronix/NodeMCU_four_relay_board/tree/master/Wifi_Four_SSR_Virtual_Switch](url)
> ARDUINO_VARIANT = nodemcuv2
> tried core v 2.3.0 as well as 2.4.0

```
arduino-makefile/Esp8266-Arduino-Makefile/esp8266-2.3.0/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: build.nodemcuv2/test.elf section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
../../../espXArduino.mk:321: recipe for target 'build.nodemcuv2/test.elf' failed
make: *** [build.nodemcuv2/test.elf] Error 1
```

swapnil823, 2018-01-25:
Update: Works with dev Branch of makefile

closed 2018-01-25 by swapnil823

thunderace, 2018-01-26:
Yes, i have to merge the dev branch in the master one. But it need some documentation and i haven't enough time for now (i'am flooded in france)