thunderace/Esp8266-Arduino-Makefile #24: ./esp8266-install.sh: 13: [: unexpected operator

flohoff, 2019-03-05:
The install of the esp8266 toolchain fails with shell error - /bin/sh is not bash but dash on modern Debian distributions - Using bash fixes the issue:

```
 12 cd esp8266-$ESP8266_VER/tools && ./get.py && cd ../..
 13 if [ "$OSTYPE" == "cygwin" ] || [ "$OSTYPE" == "msys" ]; then
 14         chmod +x ./esp8266-$ESP8266_VER/tools/esptool/esptool.exe
```

I'd suggest a change like this:

```
-if [ "$OSTYPE" == "cygwin" ] || [ "$OSTYPE" == "msys" ]; then
+if [ "$OSTYPE" = "cygwin" -o "$OSTYPE" = "msys" ]; then
        chmod +x ./esp8266-$ESP8266_VER/tools/esptool/esptool.exe
        chmod +x ./esp8266-$ESP8266_VER/tools/mkspiffs/mkspiffs.exe
 fi
```

Flo

thunderace, 2019-03-12:
Thanks for the report.
I work on an old debian 8
So, if the first line is changed from #!/bin/sh to #!/bin/bash it will be OK?

thunderace, 2019-03-13:
Tested on debian 9.8 and cygwin with #!/bin/bash -> work

commit b23f9c6a (2019-03-13): #!C:/Program Files/Git/usr/bin/bash for all install scripts in order to work with all configuration - fix #24
closed 2019-03-13 by thunderace