thunderace/Esp8266-Arduino-Makefile #27: Support For recursive libraries

swapnil823, 2019-06-01:
Some of the ESP32 libraries are dependent on other ESP32 libraries. 
An example would be SPIFFS which uses FS library. This makefile fails to recognize such usage.
Causing build to fail with following error:

`In file included from configUtils.h:6:0,
                 from configUtils.cpp:1:
../espMakefile/esp32-1.0.2/libraries/SPIFFS/src/SPIFFS.h:17:16: fatal error: FS.h: No such file or directory
compilation terminated.`

swapnil823, 2019-06-01:
Workaround as of now is to add the referenced library in USER src files so that it is included while compilation.

thunderace, 2019-06-03:
Hello, look at the SPIFFS examples : 
The first two lines are 
#include "FS.h"
#include "SPIFFS.h"

closed 2024-09-22 by thunderace