thunderace/Esp8266-Arduino-Makefile #34: Can't build for nano32

IsaacMAllen, 2024-03-30:
I'm not sure what I'm doing wrong. When I go to run make I get this error:
```
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:894: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:898: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:895: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:902: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:898: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:901: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:905: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:902: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:904: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:908: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:905: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:907: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:911: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:908: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:910: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:914: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:911: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:913: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:917: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:914: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:916: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:920: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:917: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:919: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:923: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:920: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:922: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:929: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:923: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:925: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:933: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:929: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:932: *** mixed implicit and normal rules: deprecated syntax
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:936: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:933: warning: ignoring old recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:942: warning: overriding recipe for target 'build.nano32'
/home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk:936: warning: ignoring old recipe for target 'build.nano32'
make: Circular build.nano32 <- build.nano32 dependency dropped.
[ARDUINO_LIBS] :
[USER_LIBS] :
mkdir: invalid option -- '2'
Try 'mkdir --help' for more information.
make: *** No rule to make target 'dirs', needed by 'all'.  Stop.
```

my Makefile:
```
ARDUINO_VARIANT = nano32
SERIAL_PORT = /dev/ttyACM0
ARDUINO_ARCH=esp32
# uncomment and set the right serail baud according to your sketch (default to 115200)
#SERIAL_BAUD = 115200
# uncomment this to use the 1M SPIFFS mapping
#SPIFFS_SIZE = 1
# uncomment to exclude this lib from dependencies
#EXCLUDE_USER_LIBS=ESP8266TrueRandom
include /home/isea/Esp8266-Arduino-Makefile/esp32Arduino.mk
```
the sketch file:
```
String nom = "Arduino";
String msg;

void setup() {
  Serial.begin(9600);
}

void readSerialPort() {
  msg = "";
  if (Serial.available()) {
    delay(10);
    while (Serial.available() > 0) {
      msg += (char)Serial.read();
    }
    Serial.flush();
  }
}

void sendData() {
  //write data
  Serial.print(nom);
  Serial.print(" received : ");
  Serial.print(msg);
}

void loop() {
  readSerialPort();

  if (msg != "") {
    sendData();
  }
  delay(500);
}
```

thunderace, 2024-09-22:
Sorry for my late response, i was offline for a while. On which platform? Linux?

thunderace, 2024-11-17:
Fixed in the last release

closed 2024-11-17 by thunderace